meshtastic/Meshtastic-Android #2541: [Bug]: Device names are wrong on network connection page

linker3000, 2025-07-27:
### Contact Details

linker3000@gmail.com

### What happened?

Upgraded app to latest alpha.

The names associated with IP-connected local devices are wrong, for example I have two IP-based devices but the name of one is currently listed in the app as that of my Bluetooth-only portable node. In the attached screenshot they are both incorrectly named the same.

The names displayed will swap around.

Tried: Clearing cache. Clearing all memory. Reinstalling app after clearing all memory.

### App Version

2.6.31 34956

[image]

### Phone

Motorola g85 5G 

### Device

Heltec V3 IP,  Heltec V3 (Bluetooth). Seeed ESP+Wio kit (IP), RaK Wisblock NRF (Bluetooth)

### Firmware

Not relevant 

### Relevant log output

```shell
N/R
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-07-27:
fixed by #2538 when merged

jamesarich, 2025-07-27:
actually this is a dupe of #2499

closed 2025-07-27 by jamesarich