meshtastic/Meshtastic-Android #2547: [Bug]: No Network option in Radio Configuration

netureza-pt, 2025-07-28:
### What happened?

Under "Radio Configuration" there is no "Network" option to configure WiFi or Ethernet.

Is it supposed to only show when the device has WiFi or Ethernet hardware?

My device (WisMesh Ethernet MQTT Gateway) has ethernet. When powered by USB or PoE the ethernet link is established at 100Mb/s full duplex, but there is no dhcp request and no way to configure it in the app.

I am not sure this is a problem in the app, the firmware or the hardware.

See also my [post on RAKWireless forum](https://forum.rakwireless.com/t/no-network-on-wismesh-ethernet-mqtt-gateway/15245/2)

### App Version

2.6.30

### Phone

Asus Zenphone
LG Q6

### Device

WisMesh Ethernet MQTT Gateway (with PoE)

### Firmware

2.6.11.60ec05e

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-07-28:
Likely Firmware / settings related. 

The Python log seems to indicate that it is reporting as not having ethernet.

netureza-pt, 2025-07-28:
Can you please confirm if the Network menu should be there even if the device does not have Ethernet nor WiFi?
If it should, then there is a bug in the Android app because I don't have that option on any of my devices.

If not, I will close this issue and open a new one in the firmware repository.
Or maybe the issue can be moved there?

DaneEvans, 2025-07-28:
Looking at the code excerpt, it should be hidden for hardware that doesn't
have wifi or ethernet.

Maybe it should just be disabled, as that would cause less confusion.

I think we can move it, I'll check when I get on a computer.

DaneEvans, 2025-07-29:
I don't have rights to transfer it, so it's probably easier to recreate it.