meshtastic/Meshtastic-Android #2558: [Feature Request]: Export only logs containing "seq" as payload when exporting rangetest.csv

Nikotine1, 2025-07-29:
### Tell us your idea.

Currently, when exporting rangetest.csv in the Android app, it exports all logs. When you plot these in Google Earth, you see all logs, not just the ones for the rangetest. 

It would be better if exporting rangetest.csv filtered for payloads containing "seq". 

Example of an exported csv (lat and long removed):

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-07-30:
How are you exporting that? 
So far as I can recall the only dedicated range test export is via the firmware on the ESPs. 

On the debug panel, you can export everything, but you would need to filter it elsewhere I think.

DaneEvans, 2025-08-24:
> This is good idea but should be optional. I do not like using RangeTest as
> it requires lots of reconfiguration on both fixed node and mobile node,
> especially if one wants to avoid clogging running network. I prefer simply
> to record logs on my own and export it. So, exporting rangetest.csv with
> whole log is actually useful.

Of course there's someone using a bug as a feature. It's like user 101.

I left the code in place, we've got some UI changes happening at the
moment, so once things settle, I'll get you a button to export full logs to
CSV separately.