meshtastic/Meshtastic-Android #2590: [Bug]: Autoconnect via Bluetooth (not working)

spaetz, 2025-07-31:
### What happened?

I connect via Bluetooth with my Heltec V3. When I move out of range and return, the connection to my meshtastic device is not automatically taken up again. I need to manually switch away from the bluetooth device to "deactivated" and back in order to reconnect.
Is this supposed to work?

### App Version

2.6.30

### Phone

stock Android 14

### Device

Heltec v3

### Firmware

2.6.13

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-07-31:
dupe : #2347

closed 2025-07-31 by DaneEvans

spaetz, 2025-07-31:
Ohh, i had searched for duplicates but must have missed this. I apologize.