meshtastic/Meshtastic-Android #2604: [Bug]: Regression - Device metrics log is hidden

DaneEvans, 2025-08-02:
### Contact Details

@DaneEvans

### What happened?

@jamesarich the changes to hide logs without data has taken out Device Metrics

### App Version

2.6.32

### Phone

Motog86 + sims

### Device

any

### Firmware

2.6.11

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-08-02:
Does it or does it just take a second or two to load. 
I dont see any issues on 2.6.32 + 2.7.4

b8b8, 2025-08-02:
For what its worth, it is working for me from a master T1000E on 2.7.4 remotely viewing a t-beam on 2.6.5.

DaneEvans, 2025-08-02:
@b8b8 are you on release, or HEAD? 
Currently f5478b42c36264e1392c8a0ba710f214a75779c9

DaneEvans, 2025-08-02:
I think maybe I killed it in 1ba70ca54783f5e806d4eba5d65f548272f0f97c #2556

DaneEvans, 2025-08-02:
Confirmed. Rolling back just that MR seems to have solved it.

b8b8, 2025-08-02:
That merge was a good addition. I look forward to it being fixed.

DaneEvans, 2025-08-02:
@b8b8 
It's not going anywhere, but keep an eye on any other metrics that might be being accidentally hidden by it. I have limited sensors to be exhasitive

PR #2607 (2025-08-02): Fix regression of deviceMetrics
- handle both env and dev metrics coming in in separate telemetry messages

Closes #2604 

Introduced in #2556
My concern with that one was vindicated. 
Need to keep an eye on any other metrics that may fall through the same cracks. 

And we need to get some UI testing up.
closed 2025-08-02 by DaneEvans