meshtastic/Meshtastic-Android #2605: [Bug]: Primary Chanel location is always shown as disabled even if it is enabled

RCGV1, 2025-08-02:
### What happened?

I noticed what may be a bug in the android app that is probably a easy fix. It always thinks that the position is disabled for the channel even though I have verified that in that nodes config the primary channel has position enabled

### App Version

2.6.3

### Phone

Pixel 8

### Device

T1000E

### Firmware

2.7.3

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-07:
@RCGV1 Where are you seeing this information ? Ideally with pictures. 

config > nodes doesn't exist, and apart from the lock icons I don't think we present that info directly.

RCGV1, 2025-08-07:
Nvm I have confirmed it worked as intended if I see this happen again I will reopen and provide better documentation of the issue