meshtastic/Meshtastic-Android #2609: [Bug]: app crash when interacting with node while chat reply is active

bunder2015, 2025-08-02:
### What happened?

Open chat window, click the reply arrow from any of the messages, then click the short code button for any node in the chat.  App crashes.  

### App Version

2.6.32

### Phone

S22, Android 14 (One UI 6.1)

### Device

Tdeck

### Firmware

2.7.3

### Relevant log output

```shell
App crashed, I don't think there is a log anymore. :)
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-02:
Oh, there's logs. 

Replicated 
```
2025-08-02 18:18:51.540 18755-18778 com.geeksv...hInterface com.geeksville.mesh                  D  Done reading from radio, fromradio is empty
2025-08-02 18:18:51.630 18755-19573 com.geeksv...BuildUtils com.geeksville.mesh                  W  Failed to fetch device hardware from server: empty response from server
2025-08-02 18:18:51.631 18755-19573 com.geeksv...BuildUtils com.geeksville.mesh                  D  Using stale cached device hardware
2025-08-02 18:18:53.600 19175-19181 artd                    artd                                 I  GetBestInfo: /data/app/~~Y9_ksDp1nxDU-S0mRpdttw==/com.geeksville.mesh-_HzG774tUM9DC87ZjVO74Q==/base.apk has no usable artifacts
2025-08-02 18:18:54.866 32037-8984  com.qqlabs...derService com.qqlabs.minimalistlauncher        D  Current App in foreground is: com.geeksville.mesh
2025-08-02 18:18:56.051 18755-18755 WindowOnBackDispatcher  com.geeksville.mesh                  W  OnBackInvokedCallback is not enabled for the application.
                                                                                                    Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
2025-08-02 18:18:59.869 32037-8984  com.qqlabs...derService com.qqlabs.minimalistlauncher        D  Current App in foreground is: com.geeksville.mesh
2025-08-02 18:19:00.178 18755-18755 Choreographer           com.geeksville.mesh                  I  Skipped 43 frames!  The application may be doing too much work on its main thread.
2025-08-02 18:19:04.873 32037-8984  com.qqlabs...derService com.qqlabs.minimalistlauncher        D  Current App in foreground is: com.geeksville.mesh
2025-08-02 18:19:06.188 18755-18755 com.geeksv...sViewModel com.geeksville.mesh                  D  MetricsViewModel created
2025-08-02 18:19:06.195 18755-18793 com.geeksv...BuildUtils com.geeksville.mesh                  W  Failed to fetch device hardware from server: empty response from server
2025-08-02 18:19:06.432 18755-18755 com.geeksv...BuildUtils com.geeksville.mesh                  W  Failed to fetch firmware releases from server: empty response from server
2025-08-02 18:19:06.585 18755-19597 com.geeksv...BuildUtils com.geeksville.mesh                  D  Using stale cached device hardware
2025-08-02 18:19:06.807 18755-18755 com.geeksv...iceVersion com.geeksville.mesh                  W  Exception while parsing version '', assuming version 0
2025-08-02 18:19:06.807 18755-18755 com.geeksv...iceVersion com.geeksville.mesh                  W  Exception while parsing version '', assuming version 0
2025-08-02 18:19:07.160 18755-18755 Choreographer           com.geeksville.mesh                  I  Skipped 49 frames!  The application may be doing too much work on its main thread.
2025-08-02 18:19:07.246 18755-18755 com.geeksv...iceVersion com.geeksville.mesh                  W  Exception while parsing version '', assuming version 0
2025-08-02 18:19:07.246 18755-18755 com.geeksv...iceVersion com.geeksville.mesh                  W  Exception while parsing version '', assuming version 0
2025-08-02 18:19:07.352 18755-18755 AndroidRuntime          com.geeksville.mesh                  E  FATAL EXCEPTION: main (Ask Gemini)
                                                                                                    Process: com.geeksville.mesh, PID: 18755
                                                                                                    java.lang.IllegalStateException: MutableState(value=Message(uuid=137, receivedTime=1754099978132, node=Node(num=-1597304584, metadata=firmware_version: "2.7.3.cf574c7"
                                                                                                    device_state_version: 24
                                                                                                    canShutdown: true
                                                                                                    hasWifi: true
                                                                                                    hasBluetooth: true
                                                                                                    position_flags: 811
                                                                                                    hw_model: HELTEC_V3
                                                                                                    hasPKC: true
                                                                                                    excluded_modules: 1280
                                                                                                    , user=id: "!a0cb10f8"
                                                                                                    long_name: "Heltec 10f8"
                                                                                                    short_name: "H f8"
                                                                                                    macaddr: "0\355\240\313\020\370"
                                                                                                    hw_model: HELTEC_V3
                                                                                                    public_key: "\243\354t\256\264k\222sU\241\246\225G\324\232Ts3\305~k\266\260\250\bC!\214\323}\350;"
[... cut ...]
2025-08-02 18:19:08.736 19651-19651 geeksville.mesh         pid-19651                            W  Redefining intrinsic method java.lang.Thread java.lang.Thread.currentThread(). This may cause the unexpected use of the original definition of java.lang.Thread java.lang.Thread.currentThread()in methods that have already been compiled.
2025-08-02 18:19:08.737 19651-19651 geeksville.mesh         pid-19651                            W  Redefining intrinsic method boolean java.lang.Thread.interrupted(). This may cause the unexpected use of the original definition of boolean java.lang.Thread.interrupted()in methods that have already been compiled.

```

jsoberg, 2025-08-02:
This must be a relatively new issue - tried on my version of the app that I built ~a week ago and no crash, but just rebuilt with latest and saw the crash