meshtastic/Meshtastic-Android #2627: [Bug]: Android app causing wifi disconnection

cbvicious, 2025-08-04:
### Contact Details

cbvicious on discord

### What happened?

After taking certain actions (trace route or request user info) in the Android app, it seems that the arrival of the response causes the phone's wifi connection to disconnect and reconnect. 

What did I expect to happen? Certainly not my phone disconnecting from WiFi every time it receives a response to a user action on the meshtastic app. 

https://github.com/user-attachments/assets/d294a7ad-896b-47ae-a135-b24918aad031

### App Version

2.6.32 (35009) alpha Google APK

### Phone

Pixel 6 pro

### Device

Portduino (Raspberry Pi Zero 2 W with a MeshADV Mini hat)

### Firmware

2.7.3

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

cbvicious, 2025-08-05:
Issue seems to have reduced on 2.6.33, but still exists. Traceroute no longer seems to cause the wifi crash, but the settings menu still does. Tapping into a category, back out, and then on another category hangs at 0% (for minutes on end) until the wifi crash occurs. 

I'm happy to provide lots of someone can tell me how to narrow it down from the thousands of lines including PII that are generated during the event. 

It appears that it causes MDNS and NearbyDevices to panic, but I'm not sure if that's a cause or symptom of the disconnect.

https://github.com/user-attachments/assets/e53cb890-7f08-4120-95e0-a5e36ff44167

DaneEvans, 2025-08-05:
for a start from `com.geeksville.mesh`
But you'll need to figure out what ones are generated from WiFi etc. and are needed to show the bigger picture. 

It could be that the app logs show enough, but without seeing the logs, I can't see that. 

Also need OS version info,

cbvicious, 2025-08-05:
Android 16, build number: BP31.250610.004.A1

This event generated approximately 20,000 lines of log in under a second. It's a gargantuan task without some filtering.

cbvicious, 2025-08-05:
2025-08-05 01:09:43.479 12505-12505 InsetsController        com.geeksville.mesh                  D  hide(ime())
2025-08-05 01:09:43.480 12505-12505 ImeTracker              com.geeksville.mesh                  I  com.geeksville.mesh:ceff4e80: onCancelled at PHASE_CLIENT_ALREADY_HIDDEN
2025-08-05 01:09:45.204 12505-12505 WindowOnBackDispatcher  com.geeksville.mesh                  D  setTopOnBackInvokedCallback (unwrapped): android.app.Dialog$$ExternalSyntheticLambda2@8f2d3c9
2025-08-05 01:09:45.204 12505-12505 WindowOnBackDispatcher  com.geeksville.mesh                  D  setTopOnBackInvokedCallback (unwrapped): androidx.activity.OnBackPressedDispatcher$Api34Impl$createOnBackAnimationCallback$1@2619ace
2025-08-05 01:09:45.251 12505-12505 ImeBackDispatcher       com.geeksville.mesh                  D  switch root view (mImeCallbacks.size=0)
2025-08-05 01:09:45.253 12505-12505 InsetsController        com.geeksville.mesh                  D  hide(ime())
2025-08-05 01:09:45.253 12505-12505 ImeTracker              com.geeksville.mesh                  I  com.geeksville.mesh:e0e2df4c: onCancelled at PHASE_CLIENT_ALREADY_HIDDEN
2025-08-05 01:09:57.296 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  E  IOException in TCP reader: java.net.SocketException: Software caused connection abort
2025-08-05 01:09:57.296 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  E  IOException in TCP reader: java.net.SocketException: Software caused connection abort
2025-08-05 01:09:57.297 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  D  Closing TCP socket
2025-08-05 01:09:57.302 12505-12505 com.geeksv...aceService com.geeksville.mesh                  I  stopping interface com.geeksville.mesh.repository.radio.TCPInterface@8bb8638
2025-08-05 01:09:57.302 12505-12505 com.geeksv...aceService com.geeksville.mesh                  I  stopping interface com.geeksville.mesh.repository.radio.TCPInterface@8bb8638
2025-08-05 01:09:57.302 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  D  Closing TCP socket
2025-08-05 01:09:57.305 12505-12505 com.geeksv...PInterface com.geeksville.mesh                  D  Closing stream for good
2025-08-05 01:09:57.305 12505-26180 com.geeksv...aceService com.geeksville.mesh                  D  Broadcasting connection=false
2025-08-05 01:09:57.305 12505-12505 com.geeksv...PInterface com.geeksville.mesh                  D  Closing stream for good
2025-08-05 01:09:57.305 12505-26180 com.geeksv...aceService com.geeksville.mesh                  D  Broadcasting connection=false
2025-08-05 01:09:57.306 12505-26162 com.geeksv...eshService com.geeksville.mesh                  D  onConnectionChanged: CONNECTED -> DISCONNECTED
2025-08-05 01:09:57.306 12505-26162 com.geeksv...eshService com.geeksville.mesh                  D  onConnectionChanged: CONNECTED -> DISCONNECTED
2025-08-05 01:09:57.306 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  D  Reconnect attempt 1 in 1s
2025-08-05 01:09:57.306 12505-26180 com.geeksv...PInterface com.geeksville.mesh                  D  Reconnect attempt 1 in 1s
2025-08-05 01:09:57.306 12505-26162 com.geeksv...eshService com.geeksville.mesh                  I  Stopping packet queueJob
2025-08-05 01:09:57.306 12505-26162 com.geeksv...eshService com.geeksville.mesh                  I  Stopping packet queueJob
2025-08-05 01:09:57.307 12505-26175 com.geeksv...eshService com.geeksville.mesh                  D  queueJob packet id=2154199791 success false
2025-08-05 01:09:57.307 12505-26175 com.geeksv...eshService com.geeksville.mesh                  D  queueJob packet id=2154199791 success false
2025-08-05 01:09:57.308 12505-26162 com.geeksv...eAnalytics com.geeksville.mesh                  D  Analytics: track mesh_disconnect
2025-08-05 01:09:57.308 12505-26162 com.geeksv...eAnalytics com.geeksville.mesh                  D  Analytics: track mesh_disconnect
2025-08-05 01:09:57.308 12505-26162 com.geeksv...eAnalytics com.geeksville.mesh                  D  Analytics: track num_nodes
2025-08-05 01:09:57.308 12505-26162 com.geeksv...eAnalytics com.geeksville.mesh                  D  Analytics: track num_nodes
2025-08-05 01:09:57.351 12505-12505 WindowOnBackDispatcher  com.geeksville.mesh                  D  setTopOnBackInvokedCallback (unwrapped): android.app.Activity$$ExternalSyntheticLambda0@7f0108d
2025-08-05 01:09:57.356 12505-12505 com.geeksv...gViewModel com.geeksville.mesh                  D  RadioConfigViewModel created
2025-08-05 01:09:57.356 12505-12505 com.geeksv...gViewModel com.geeksville.mesh                  D  RadioConfigViewModel created
2025-08-05 01:09:57.356 12505-12505 com.geeksv...TScanModel com.geeksville.mesh                  D  BTScanModel created
2025-08-05 01:09:57.356 12505-12505 com.geeksv...TScanModel com.geeksville.mesh                  D  BTScanModel created
2025-08-05 01:09:57.419 12505-12505 com.geeksv...aceService com.geeksville.mesh                  I  Starting radio ...227
2025-08-05 01:09:57.419 12505-12505 com.geeksv...aceService com.geeksville.mesh                  I  Starting radio ...227
2025-08-05 01:09:57.419 12505-26194 com.geeksv...PInterface com.geeksville.mesh                  D  TCP connecting to 192.168.1.227
2025-08-05 01:09:57.419 12505-26194 com.geeksv...PInterface com.geeksville.mesh                  D  TCP connecting to 192.168.1.227
[... cut ...] bjects, 71% free, 38MB/134MB, paused 399us,5.178ms total 99.899ms
2025-08-05 01:09:58.071 12505-12505 ImeBackDispatcher       com.geeksville.mesh                  D  Clear (mImeCallbacks.size=0)
2025-08-05 01:09:58.071 12505-12505 ImeBackDispatcher       com.geeksville.mesh                  D  switch root view (mImeCallbacks.size=0)
2025-08-05 01:09:58.071 12505-12505 WindowOnBackDispatcher  com.geeksville.mesh                  D  setTopOnBackInvokedCallback (unwrapped): null
2025-08-05 01:09:58

cbvicious, 2025-08-05:
And the android system log around that TCP Reader error from com.geeksville.mesh 

[logcat.txt](https://github.com/user-attachments/files/21593086/logcat.txt)