meshtastic/Meshtastic-Android #2636: [Bug]: Pairing Heltec v3 to Samsung S23

DustinTrib, 2025-08-05:
### Contact Details

Discord DustinT's

### What happened?

A bug happened! I have 4 Heltec v3 nodes. All 4 of them have been working on my Samsung S23 for months. 

**What I was doing when the issue occurred**
About a week ago 7/28 I wanted to move one of the nodes to one of my kids to spread the nodes out a bit. I forgot (Unpaired) the node from my Samsung and paired it to his iPhone. Pairing was odd but still paired after a couple of tries. Later, on my Samsung, I noticed that one of my remaining nodes wouldn't connect even though it was showing in the list of nodes. Two others would still connect until I stupidly unpaired them too. Now none of my nodes will pair to my Samsung.

https://github.com/user-attachments/assets/54e56937-9ad2-48e7-849f-2151784a8b06

 Listed below are the details and steps taken to reconnect my nodes.

Versions:
- Heltec v3's Firmware 2.6.11.60ec05e Beta
- Samsung S23 Ultra Factory Android 15 Security patch level July 1, 2025 and last update July 29, 2025
Steps taken to resolve the issue:
- Phone Forget (Unpair) all nodes. The nodes do NOT appear in the BT list
- Phone Rebooted several times
- Heltec v3 Delete and reflashed the node with 2.5.20.x Beta and 2.7.3.x Alpha with no success so I reflashed it back to 2.6.11.x Beta
- iPad Paired the node to my iPad. Pairing was flaky but it was successful
- iPad Forgot the node and deleted Meshtastic from the iPad because I want it on my Samsung
- Serial I connected the node to my Samsung over serial, but no node configuration was successful

### App Version

2.6.30

### Phone

Samsung s23 Ultra, Android 15

[image]

### Device

heltec v3

### Firmware

2.6.11.60ec05e Beta

### Relevant log output

```shell
8/4/25 6:48:44 PM [Packet]
from: 862412364 (!33675e4c)
to: 4294967295 (!ffffffff)
decoded {
  portnum: TELEMETRY_APP
  payload: "\r\333F\221h2 \by\025\366(\374?\035\364\266,= \002(\0020\0018\001@\004H\001`\200\335\021h\360\370\a"
}
id: 1250994231
rx_time: 1754351323
hop_limit: 3
priority: BACKGROUND

8/4/25 6:48:44 PM [Packet]
from: 862412364 (!33675e4c)
to: 4294967295 (!ffffffff)
decoded {
  portnum: TELEMETRY_APP
  payload: "\r\333F\221h\022\023\bM\025\006\201}@\035\366(\374?%\364\266,=(y"
}
id: 721824822
rx_time: 1754351323
hop_limit: 3
priority: BACKGROUND

8/4/25 6:47:43 PM [Packet]
from: 862412364 (!33675e4c)
to: 4294967295 (!ffffffff)
decoded {
  portnum: TELEMETRY_APP
  payload: "\r\236F\221h\022\023\bI\025\252\361z@\035\217\302E@%\253\361\322<(="
  bitfield: 0
}
id: 2831647797
rx_time: 1754351262
hop_limit: 3
priority: BACKGROUND
hop_start: 3
relay_node: 76

8/4/25 6:47:16 PM [Packet]
from: 862412364 (!33675e4c) (!33675e4c)
to: 862412364
decoded {
  portnum: ROUTING_APP
  payload: "\030\000"
  request_id: 2154019891
}
id: 1116068916
rx_time: 1754351235
hop_limit: 3
priority: ACK

8/4/25 6:47:15 PM [Packet]
to: 862412364 (!33675e4c)
decoded {
  portnum: ADMIN_APP
  payload: "\335\0 (!00000000)02\203F\221h"
}
id: 2154019891
hop_limit: 3
want_ack: true
priority: RELIABLE

8/4/25 6:47:15 PM [ConfigComplete]
2

8/4/25 6:47:15 PM [NodeInfo]
num: 862155888 (!33637470)
user {
  id: "!33637470"
  long_name: "Meshtastic 7470"
  short_name: "7470"
  macaddr: "d\3503ctp"
  hw_model: HELTEC_V3
  public_key: "\034(\253-kO\257f\023\006m\262\276Y\336wb\212#\252\300lM]\214\370\354\233\002\352\201`"
  is_unmessagable: false
}
snr: 6.25
last_heard: 1753921720
hops_away: 0

8/4/25 6:47:15 PM [NodeInfo]
num: 862415700 (!33676b54)
user {
  id: "!33676b54"
  long_name: "PH70"
  short_name: "PH70"
  macaddr: "d\3503gkT"
  hw_model: HELTEC_V3
  public_key: "hRTq\220\370\017\332\276\374\274\312|\315Y\024\000`\201vCD\021\221o4\315\034\241Ib7"
  is_unmessagable: false
}
snr: -3.0
last_heard: 1753919649
device_metrics {
  battery_level: 101
  voltage: 4.282
  channel_utilization: 0.0
  air_util_tx: 0.1268889
  uptime_seconds: 90789
}
hops_away: 0

8/4/25 6:47:15 PM [NodeInfo]
num: 862414064 (!336764f0)
user {
  id: "!336764f0"
  long_name: "jeepjeep"
  short_name: "jeep"
  macaddr: "d\3503gd\360"
  hw_model: HELTEC_V3
  public_key: "\'X\000\277(\371wr\346\354\225\316\v\rf\3737\275i\033V\211\305\254o`.\350\r\250\351/"
  is_unmessagable: false
}
snr: 5.75
last_heard: 1753398381
device_metrics {
  battery_level: 47
  voltage: 3.697
  channel_utilization: 0.0
  air_util_tx: 0.13022223
  uptime_seconds: 6546
}
hops_away: 0

8/4/25 6:47:15 PM [ModuleConfig PAXCOUNTER]
paxcounter {
}

8/4/25 6:47:15 PM [ModuleConfig DETECTION_SENSOR]
detection_sensor {
  minimum_broadcast_secs: 45
  detection_trigger_type: LOGIC_HIGH
}

8/4/25 6:47:15 PM [ModuleConfig AMBIENT_LIGHTING]
ambient_lighting {
  current: 10
  red: 103
  green: 94
  blue: 76
}

8/4/25 6:47:15 PM [ModuleConfig NEIGHBOR_INFO]
neighbor_info {
}

8/4/25 6:47:15 PM [ModuleConfig REMOTE_HARDWARE]
remote_hardware {
}

8/4/25 6:47:15 PM [ModuleConfig AUDIO]
audio {
}

8/4/25 6:47:15 PM [ModuleConfig CANNED_MESSAGE]
canned_message {
}

8/4/25 6:47:15 PM [ModuleConfig TELEMETRY]
telemetry {
}

8/4/25 6:47:15 PM [ModuleConfig RANGE_TEST]
range_test {
}

8/4/25 6:47:15 PM [ModuleConfig STORE_FORWARD]
store_forward {
}

8/4/25 6:47:15 PM [ModuleConfig EXTERNAL_NOTIFICATION]
external_notification {
}

8/4/25 6:47:15 PM [ModuleConfig SERIAL]
serial {
}

8/4/25 6:47:15 PM [ModuleConfig MQTT]
mqtt {
  address: "mqtt.meshtastic.org"
  username: "meshdev"
  password: "large4cats"
  encryption_enabled: true
  root: "msh/US"
}

8/4/25 6:47:15 PM [Config DEVICE_UI]
device_ui {
}

8/4/25 6:47:15 PM [Config SESSIONKEY]
sessionkey {
}

8/4/25 6:47:15 PM [Config SECURITY]
security {
  public_key: "zV\363\304Q\314\203h\255{\367\310\245\265\032\201s#qj\312\271\003\aq\0163\360g\031:a"
  private_key: "\2009Iy\241\347\365\354D\376\374\372\271*\b{\227\344r\337\313V\250y\210\351\344\226+\371~F"
  serial_enabled: true
}

8/4/25 6:47:15 PM [Config BLUETOOTH]
bluetooth {
  enabled: true
  fixed_pin: 123456
}

8/4/25 6:47:15 PM [Config LORA]
lora {
  use_preset: true
  region: US
  hop_limit: 3
  tx_enabled: true
  tx_power: 30
  sx126x_rx_boosted_gain: true
}

8/4/25 6:47:14 PM [Config DISPLAY]
display {
  screen_on_secs: 60
  units: IMPERIAL
}

8/4/25 6:47:14 PM [Config NETWORK]
network {
  ntp_server: "meshtastic.pool.ntp.org"
  enabled_protocols: 1
}

8/4/25 6:47:14 PM [Config POWER]
power {
  wait_bluetooth_secs: 60
  sds_secs: 4294967295
  ls_secs: 300
  min_wake_secs: 10
}

8/4/25 6:47:14 PM [Config POSITION]
position {
  position_broadcast_secs: 900
  position_broadcast_smart_enabled: true
  gps_update_interval: 120
  position_flags: 811
  broadcast_smart_minimum_distance: 100
  broadcast_smart_minimum_interval_secs: 30
  gps_mode: NOT_PRESENT
}

8/4/25 6:47:14 PM [Config DEVICE]
device {
  role: CLIENT_MUTE
  node_info_broadcast_secs: 10800
}

8/4/25 6:47:14 PM [Channel]
index: 7
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 6
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 5
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 1
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 3
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 4
settings {
}

8/4/25 6:47:14 PM [Channel]
index: 2
settings {
}

8/4/25 6:47:14 PM [Channel]
settings {
  psk: "\001"
  module_settings {
    position_precision: 15
  }
}
role: PRIMARY

8/4/25 6:47:14 PM [DeviceMetadata]
firmware_version: "2.6.11.60ec05e"
device_state_version: 24
canShutdown: true
hasWifi: true
hasBluetooth: true
role: CLIENT_MUTE
position_flags: 811
hw_model: HELTEC_V3
hasPKC: true
excluded_modules: 1280

8/4/25 6:47:14 PM [NodeInfo]
num: 862412364 (!33675e4c)
user {
  id: "!33675e4c"
  long_name: "mdtr"
  short_name: "mdtr"
  macaddr: "d\3503g^L"
  hw_model: HELTEC_V3
  role: CLIENT_MUTE
  public_key: "zV\363\304Q\314\203h\255{\367\310\245\265\032\201s#qj\312\271\003\aq\0163\360g\031:a"
  is_unmessagable: false
}
position {
}
device_metrics {
  battery_level: 62
  voltage: 3.822
  channel_utilization: 3.09
  air_util_tx: 0.025750002
  uptime_seconds: 61
}
is_favorite: true

8/4/25 6:47:14 PM [MyNodeInfo]
my_node_num: 862412364 (!33675e4c)
reboot_count: 254
min_app_version: 30200
device_id: "\345Z\f7/C+M\342\350\323\017]\242\300\210"
pio_env: "heltec-v3"
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-07:
It would help 

Android studio -> logcat 
have a look and see if you can spot anything. 

Filters com.geeksville.mesh , and try and find whatever bluetooth services you can find. 

Try to avoid posting any personal info.

Danie10, 2025-08-07:
I'm watching this issue as I've spent a week struggling with Bluetooth between my T-Echo and my Samsung S23 Ultra - it will pair but struggles to read the T-Echo and keeps losing the Bluetooth connection although it is paired. Also on 2.6.11.60ec05e Beta. Pairs OK to Heltec V3 WL nodes but via WiFi.

DustinTrib, 2025-08-08:
DaneEvans
I am a Android Studio / logcat infant. I have spent a little time today trying to figure out how to get logcat without building an app. I don't know if it is the newer version of Andoid Studio or not. If you have any tips, I will gladly take them. I will also look for an older version of Android Studio, because that may be my problem.

Thanks in advance

jamesarich, 2025-08-17:
> > should be improved in [`v2.6.34` (release)](https://github.com/meshtastic/Meshtastic-Android/releases/tag/v2.6.34)
> 
> [@jamesarich](https://github.com/jamesarich) When will the beta be available on the Play Store in the beta program?

when it's determined stable by devs and testers

Danie10, 2025-08-17:
Certainly not better yet for me on v2.6.34(35082) of the Android app. Exactly the same issue. But hopefully the v2.6.34 release is later than the version of v.2.6.34 that I have.

jamesarich, 2025-08-17:
> Certainly not better yet for me on v2.6.34(35082) of the Android app. Exactly the same issue. But hopefully the v2.6.34 release is later than the version of v.2.6.34 that I have.

2.6.34-alpha is still an alpha version, there hasn't been another version pushed out to open testing yet - if you want a stable release, the latest available stable build is 2.6.30-beta.

Ccz119, 2025-08-17:
I have also used ver 2.6.34 and i can confirm that bluetooth is still messing itself. Does not reconnect , disconnects for no reason( dedicated phone, no nonsense) and keeps asking pass every time. Tried several boards, several phones, done resets Everywhere, but no avail.

Ccz119, 2025-08-17:
I dont get why this issue is now marked as closed. Is the bug fixed? Why not kept open, maybe someone has an idea on a solution?