meshtastic/Meshtastic-Android #2681: [Bug]: Message content is not considered sensitive by Android

rohanssrao, 2025-08-10:
### Contact Details

rohanssrao@gmail.com

### What happened?

When I tell Android to [hide sensitive notification content on the lock screen](https://support.google.com/android/answer/9079661?hl=en#zippy=%2Coption-hide-sensitive-content-from-notifications-on-your-lock-screen), Meshtastic message content is not hidden. It seems like the app does not make message content as sensitive.

### App Version

2.6.30

### Phone

Pixel 9 Pro XL, GrapheneOS (Android 15)

### Device

N/A

### Firmware

N/A

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-08-10:
https://developer.android.com/reference/android/app/Notification#VISIBILITY_SECRET

jamesarich, 2025-08-11:
This would then be user configurable at the Android system level. 

[image]