meshtastic/Meshtastic-Android #2682: [Bug]: Map Reporting Send Greyed out

voycey, 2025-08-11:
### What happened?

I have tried every combination of settings to get map reporting working on my T1000-E.
It is connected via Bluetooth and has Proxy to Client enabled

[image]

But no matter what I change the Send button remains greyed out.

### App Version

2.6.30 (latest Play store version)

### Phone

Galaxy s24+

### Device

T-1000E

### Firmware

2.6.4.b89355f

### Relevant log output

```shell
I can't get logs right now
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-08-11:
You need a minimum firmware version of 2.73 for the map report to work with the proxy and a T1000-E

closed 2025-08-11 by garthvh

voycey, 2025-08-11:
Got it, is that documented? I couldn't see it in the docs but will upgrade tonight!
Thanks!

daschu117, 2025-08-21:
I've got a Station G2 on WiFi, with Proxy to Client disabled, and the Send button is permanently grayed out if I'm trying to enable Map Reporting.