meshtastic/Meshtastic-Android #2689: [Bug]:  Crowdin translations do not change after updating the Android app

livingdeath72, 2025-08-12:
### What happened?

I have improved and corrected many translations (fi/FI) later on in the Crowdin platform, but they don't appear in the Android app even after updating it.

I've been waiting for several months and wondering why this is happening.

It would be good if the entire translation database was always reviewed, not just the newly added translations.

Here are a few examples:

Original English → now in the app → later corrected in Crowdin app

• Gas Resistance → Kaasun vastus → Kaasuvastus
• Traceroute Log → Reitinselvitys loki → Reitinselvityksen loki
• NodeDB reset → Tyhjennä Nodedb tietokanta → Tyhjennä NodeDB-tietokanta

I have also tried deleting old bad translations and replacing them with new ones and still the changes are not visible in the Android app.

There are probably more of these, but I can't remember all of them.

The Meshtastic Android app version was updated yesterday 2.6.33 to 2.6.34 and nothing happened

Here's one example.

[image]

### App Version

2.6.25 → 2.6.26 → 2.6.27 → 2.6.28 → 2.6.29 → 2.6.30 → 2.6.31 → 2.6.32 → 2.6.33 → 2.6.34

### Phone

OnePlus Nord CE 3 Lite (OxygenOS 15.0)
OnePlus 6 (Android version 11)

### Device

LILYGO® TTGO T-Beam
Heltec V3

### Firmware

Meshtastic Firmware 2.7.4.c1f4f79 Alpha
Meshtastic Firmware 2.7.3.cf574c7 Alpha

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-08-12:
> > you were also instructed in discord several times to create this issue 🙃 - some miscommunication there
> 
> Now I can report the problem correctly, but I can't do anything to make the corrected translations take effect.

One of the tasks needed here is we'll need to look at our l[ocal config](https://github.com/meshtastic/Meshtastic-Android/blob/main/config/crowdin/crowdin.yml) vs [crowdin documentation](https://support.crowdin.com/developer/configuration-file/). If you could compare those to your issue that'd be helpful.

livingdeath72, 2025-08-13:
> One of the tasks needed here is we'll need to look at our [local crowdin config](https://github.com/meshtastic/Meshtastic-Android/blob/main/config/crowdin/crowdin.yml) vs [crowdin documentation](https://support.crowdin.com/developer/configuration-file/). If you could compare those to your issue that'd be helpful.

How do I do this?

I have translated 100% of my English to Finnish and all new translations are valid, but a few corrected translations are not updated with version updates.

Would it help if someone else deleted my translations and tried to redo them?

This is a strange problem and it would be great if a solution could be found.

DaneEvans, 2025-08-13:
Listen and try to understand the information you are given. 

There are several steps in the chain to update those values, and it seems to be working on many other translations.
- [x] Make changes on crowdIn
   - [x] they persist / are accepted / shown to others
- [ ] They propogate back to here via the GH actions 
    - [ ] They are accepted
- [x] They remain in our repo
- [x] The app gets updated with new values

The Finnish file was last modified 2 weeks ago, so it's not just broken. 

I've verified that CrowdIn seems to hold the correct values (according to you, I can't comment on that). 

So we maybe have a config issue, a desync issue, or something else that we haven't thought of. 

James is suggesting comparing the example config / documentation to the one that we use, and see if anything seems wrong about the setup, eg. not overriding existing values, or anything else that may explain these symptoms

closed 2025-08-13 by jamesarich