meshtastic/Meshtastic-Android #2793: [Bug]: buggy animation

cracky22, 2025-08-20:
### Contact Details

martinblieninger2208@gmail.com

### What happened?

Hi, sorry that uploading the first video didn't work (#2764). I understand that you don't want to download a video from a file-sharing service @DaneEvans.

The bugs I found are that the cancel button doesn't work in the channel settings, and the animation laags completely when you disable the exact location and the slider extends.

https://github.com/user-attachments/assets/1f731465-995f-4502-96e8-39fef3e47223

### App Version

2.6.35 (35131) google

### Phone

Samsung Galaxy S24 Plus

### Device

Seed Tracker T1000-E

### Firmware

2.7.5.ddd1499

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-20:
Cancel is actually a 'reset changes' (did we change that string in english? Or was it used in too many places and too much effort to find them all? ) 
Use the back arrow /swipe to go back. 

The animation, do you have developed options turned on? (Spam press something in settings, you'll know if you've done it) You can change animation speeds there. Not saying you should, just that it may be a cause.  

I haven't noticed it before, but I'll see if it's reproducible in the morning

DaneEvans, 2025-08-20:
For the cancel vs clear one, jump on crowdIn and provide an update for that language.