meshtastic/Meshtastic-Android #2830: [Bug]: Heltec E290 - Factory Reset and Node Reset from the Android App Swapped

Octane14, 2025-08-24:
### Category

Hardware Compatibility

### Hardware

Heltec Vision Master E290

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11

### Description

When sending the "Factory Reset" command from the Android Meshtastic app, it appears to actually send the "NodeDB Reset" command, and vice versa.  Android App version is 2.6.35 (35131).

### Relevant log output

```Shell

```

DaneEvans, 2025-08-25:
'appears to' 
Logs / traces / whatever else you have please.

jamesarich, 2025-08-25:
Yeah i'm not seeing any changes here that would impact this - code is still sending the correct messages.

closed 2025-08-25 by Octane14