meshtastic/Meshtastic-Android #2873: [Feature Request]: Show status info on Exchange position and Exchange user info

pedjas, 2025-08-27:
### Tell us your idea.

When I click on Exchange position and Exchange user info in application and monitor traffic on other node and I see packets properly sent.

But, in application itself there are no visual clues. It is like nothing happened.

It would be good to have popup like with Traceroute to show what is sent, and if sending info went through.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

prokrypt, 2025-08-28:
Idea: in the node details page, have the Exchange position/userinfo and Request Metadata buttons do a short timeout to indicate a successful activation and to prevent accidental double-taps. For the regular pop-up menu there is no need for indication as the menu disappears when an action is selected.

jamesarich, 2025-08-29:
> > [#2740](https://github.com/meshtastic/Meshtastic-Android/pull/2740) Adresses this for metadata.
> > I'm still not convinced by this at all.
> > 2s transmission time, 2 ( 7 hops x 2 + retries (0..2) ) + network congestion etc. = 28-32 + congestion S delays
> > Unless we don't block UI, and interrupt when the packet is recieved, adding these will make for an awful UX
> 
> I am not sure to understand. I am not proposing additional traffic, just visual clue that process is executed.

It without the actual firmware support, we would be blocking ui falsely for an arbitrary timeout for no real purpose -  the packet is sent, if you get a response - the ui updates (access to remote admin). if you don't it doesn't. This isn't a hard concept.

DaneEvans, 2025-08-30:
Really? So interrupts, multiple threads? none of that rings a bell? 
Have you tried traceroute on a long (2+ hop ) route? 

The realities of a slow network make it a poor idea to block UI waiting for return packets. 
Putting a cooldown on the button, that we could do. But waiting for a response is going to be like working on a 9k dialup connection.