meshtastic/Meshtastic-Android #2932: [Feature Request]: Support for radiation telemetry property

bogdanvasut, 2025-08-30:
### Tell us your idea.

I'd like the app to display the environmentMetrics.radiation, when available, in the Environment Metrics Log.
I attach a dummy set of telemetry values injected in one node and received by another (to verify that the value was sent to the mesh).
Temperature, humidity and pressure are visible in the app, radiation is not.
Version is  2.6.30, latest in play store.
Thank you

### Relevant log output

```shell
# meshtastic --host 192.168.1.94 --listen 2>&1 | grep  -i radiation

DEBUG file:__init__.py _onTelemetryReceive line:200 in _onTelemetryReceive() asDict:{'from': 2224706868, 'to': 4294967295, 'decoded': {'portnum': 'TELEMETRY_APP', 'payload': b'\x1a\x1a\r\x00\x80&D\x15\x00\x00XB\x1d\x00 |D%\x00\x00PA\x95\x01\x00\x00PA', 'bitfield': 1, 'telemetry': {'environmentMetrics': {'temperature': 666.0, 'relativeHumidity': 54.0, 'barometricPressure': 1008.5, 'gasResistance': 13.0, 'radiation': 13.0}, 'raw': environment_metrics {
  radiation: 13
DEBUG file:__init__.py _onTelemetryReceive line:224 updating environmentMetrics metrics for 2224706868 to {'temperature': 666.0, 'relativeHumidity': 54.0, 'barometricPressure': 1008.5, 'gasResistance': 13.0, 'radiation': 13.0}
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DaneEvans, 2025-08-30:
Has Proto support
[image]

I'm leaning towards making sure everything is logged, even if we don't plot it or add a chip to details

b8b8, 2025-08-30:
It has been suggested before that another graph is made for "hazards" which would include: radiation, CO2, particulate matter, and VOC.

DaneEvans, 2025-08-31:
@b8b8 So radiation is on the details page? What about logs ? 

Or is this just a complaint about nodelist (won't happen) and plots (won't happen until I fix all my other issues with those)?

b8b8, 2025-08-31:
It's in the node details, but not the logs.

Adding it to the logs would be great. No complaint, just we are getting enough "hazards" now that maybe they warrant their own graph.

PR #2993 (2025-09-06): Feat/2932 env metrics radiation
closes #2932 

add radiation to the env logs. 

- General cleanup so all elements use the same params. 
- reduce emply space, 
- group elements together
- add preview. 

Fullly populated: 
[image]
closed 2025-09-06 by DaneEvans