meshtastic/Meshtastic-Android #2941: [Feature Request]: Separate Ack Icon for Unattended Nodes

infered5, 2025-09-01:
### Contact Details

infered5 on Discord

### Tell us your idea.

Add a new icon in public channels that differentiate if a message is acked by an unattended node, if known by the app's nodedb, so we can tell at a glance if a message was seen by a person vs static infrastructure.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

infered5, 2025-09-01:
> I'm not following the use case here - an `unmessageable` node still acks/forwards on the packet if it is a role that supports that.

The use case would help see if a message was seen by a person and you're waiting on a reply, or just seen by a nearby infrastructure node and didn't actually get seen by someone. I can send a "hello" and have it picked up by a nearby infra, but I don't actually know if it was seen by a person unless they respond, or I do some cat and mouse info exchanging on my nodelist.

Perhaps my ask was too specific. Maybe the ability to click the ack icon in the channel and get a pop-up to see which nodes have acked it? This would be a much nicer UX than diving into the debug screen.

thebentern, 2025-09-04:
The unmessagable flag on User overlaps with this idea substantially. It's unlikely we would ever add a specific ACK flavor for unattended nodes. Most ACKs are just implicit rebroadcasts.