meshtastic/Meshtastic-Android #3128: [Bug]: No Local Stats

shortwavesurfer2009, 2025-09-17:
### Checklist

- [x] I am able to reproduce the bug with the latest version.

- [x] I made sure that there are no existing **OPEN or CLOSED issues** which I could contribute my information to.

- [x] I have taken the time to fill in all the required details. I understand that the bug report will be dismissed otherwise.

- [x] This issue contains only one bug.

- [x] I have read and understood the **Contribution Guidelines**.

- [x] I agree to follow this project's Code of Conduct

### Affected app version

2.7.0-open.0

### Affected Android version

LineageOS 22.2

### Affected phone model

OnePlus Nord N200

### Affected node model

multiple: Heltec MeshPocket and Seeed T1000-E

### Affected node firmware version

2.7.7

### Steps to reproduce the bug

1. Open app and connect to a node
2. Swipe app out of app switcher so notification remains
3. Wait for notification to show stats such as uptime, airTXUtil, rx, tx, etc (15 mins max
4. Open app either by tapping notification or launcher icon
5. Pull down notification shade
6. Notification now reads example "Connected 3 online No Local Stats"

### Actual behavior

The notification reads no local stats.

### Expected behavior

The local stats should remain.

### Screenshots/Screen recordings

[image]

### Relevant log output

```shell

```

### Additional information

If you wait another 15 minutes or so, these stats show up again. But as soon as the app is opened up again, the same thing occurs.

jamesarich, 2025-09-17:
Also i believe there were some issues with 2.7.7 fw, please confirm on a stable fw.

shortwavesurfer2009, 2025-09-17:
> Also i believe there were some issues with 2.7.7 fw, please confirm on a stable fw.

2.8 was revoked due to heap growth. The stats didn't disappear when opening the app in the past. I will downgrade the T1000-E to 2.6.11 and give it a try since that is the last stable fw version.

shortwavesurfer2009, 2025-09-17:
Okay, yeah. App version 2.6.30 does not do that.

jamesarich, 2025-09-17:
thanks for doing the legwork, helps us narrow things down.

shortwavesurfer2009, 2025-09-17:
> thanks for doing the legwork, helps us narrow things down.

Okay, it works as of 2.6.33 as well. Going to try 2.6.35 next. And see if it works there.

I was playing with the 2.7 closed versions at some point, and I feel like I remember the icon changing from a green to a blue. I don't remember if that's when it started or not, but it might be closer.

shortwavesurfer2009, 2025-09-17:
I believe it to have started in 2.6.35. 2.6.33 worked, and 2.6.34 also worked. But 2.6.35 began giving me issues, so I think that's probably where it happened.

Stephan-2025, 2025-09-18:
> I believe it to have started in 2.6.35. 2.6.33 worked, and 2.6.34 also worked. But 2.6.35 began giving me issues, so I think that's probably where it happened.

Yes, i can confirm. 
I have that problems also on 2.6.35, in the versions before it was fine

ProjectMoon, 2025-09-20:
When I was testing the release candidate for app version 2.7, I would see the stats in the notification, but tapping the notification would make them disappear and say "No local stats" until the stats repopulated later.