meshtastic/Meshtastic-Android #3156: [Bug]: 2.7.0 Keeps asking to set the region

RichieFrame, 2025-09-20:
### Checklist

- [x] I am able to reproduce the bug with the latest version.

- [x] I made sure that there are no existing **OPEN or CLOSED issues** which I could contribute my information to.

- [x] I have taken the time to fill in all the required details. I understand that the bug report will be dismissed otherwise.

- [x] This issue contains only one bug.

- [x] I have read and understood the **Contribution Guidelines**.

- [x] I agree to follow this project's Code of Conduct

### Affected app version

2.7.0

### Affected Android version

Android 14

### Affected phone model

Sony Xperia XQ-EC72

### Affected node model

Any

### Affected node firmware version

2.7.9

### Steps to reproduce the bug

Disconnect node
Close app (terminate)
Open app
Reconnect node

### Actual behavior

When connecting to the node, any node, of misc firmware versions between 2.5.21 and 2.7.9, the app will throw a popup asking to set the region, which is already set on all nodes. Clicking go brings app to LoRa config page which correctly shows region set.

### Expected behavior

When region is set it should not ask you to set it.

### Screenshots/Screen recordings

[image]

### Relevant log output

```shell

```

b8b8, 2025-09-20:
Do you have a screenshot and/or log showing it's a problem on 2.7.9?

I know it was an issue earlier when the new app uses the old firmware but appears (to me) to be resolved in 2.7.9 firmware.

RichieFrame, 2025-09-20:
It was displayed as 2.7.9, but I bet in reality it was built from dev and said 2.7.9 but was actually 2.7.8 code. Tonight I finished flashing 2.7.10 dev builds onto all 2.7.x devices. There is a separate issue I have not opened an issue for where the app displays the wrong firmware version...

I had 2 devices on 2.7.8 (not sure if release or dev), 3 on 2.7.9 (early dev), 2 on 2.5.21, and 2 on 2.5.8, all showed the notice prior to tonights upgrade I am fairly sure, now, none of the 2.7.10 devices I have connected to show the notice.

frewsxcv, 2025-09-20:
This is happening to me as well. Node is on 2.7.9.