meshtastic/Meshtastic-Android #3158: [Bug]: 2.7.0 Prior connected device firmware version shown

RichieFrame, 2025-09-20:
### Checklist

- [x] I am able to reproduce the bug with the latest version.

- [x] I made sure that there are no existing **OPEN or CLOSED issues** which I could contribute my information to.

- [x] I have taken the time to fill in all the required details. I understand that the bug report will be dismissed otherwise.

- [x] This issue contains only one bug.

- [x] I have read and understood the **Contribution Guidelines**.

- [x] I agree to follow this project's Code of Conduct

### Affected app version

2.7.0

### Affected Android version

Android 14

### Affected phone model

Sony Xperia XQ-EC72

### Affected node model

Multiple

### Affected node firmware version

Multiple

### Steps to reproduce the bug

Connect to a node
Connect to a node with a different firmware version

### Actual behavior

The top connected device firmware is correct, the bottom connected radio firmware is not. In some cases a firmware upgrade notice is visible which may or may not be relevant to the actual device firmware. This was seen when testing node upgrades and the reason was not clear, with repetition it was identified the bottom firmware version was that of the previously connected radio, as it has the device model listed.

### Expected behavior

Firmware displayed will match

### Screenshots/Screen recordings

[image]
[image]
[image]

### Relevant log output

```shell

```

RichieFrame, 2025-09-20:
During the testing of the above, the device firmware as as follows:

Jackrabbit = t-echo 2.7.9
Mongoose = t-echo 2.7.10
h44c_c44c = rak4631 2.5.8 (only shown as previously connected radio)

first screenshot = h44c_c44c to Jackrabbit
second screenshot = Mongoose to Jackrabbit
third screenshot = Jackrabbit to Mongoose

Jackrabbit was upgraded to 2.7.10 after screenshots were taken and reconnection (Mongoose to Jackrabbit) shows correct firmware as expected, the top firmware version is always correct.

b8b8, 2025-09-20:
Normally switching between tabs fixes the bottom firmware field being out of sync.
Right now the plan is to actually remove the bottom field and just keep the top one next to the disconnect button.
The bottom location become redundant a bit ago and made more sense when the app version was next to (also moved).