meshtastic/Meshtastic-Android #3248: [Bug]: Active device is not shown during initial node loadinf

TapGhoul, 2025-09-29:
### Contact Details

tapghoul on discord

### Checklist

- [x] I am able to reproduce the bug with the latest version.

- [x] I made sure that there are no existing **OPEN or CLOSED issues** which I could contribute my information to.

- [x] I have taken the time to fill in all the required details. I understand that the bug report will be dismissed otherwise.

- [x] This issue contains only one bug.

- [x] I have read and understood the **Contribution Guidelines**.

- [x] I agree to follow this project's Code of Conduct

### Affected app version

2.7.1 (29314197) google

### Affected Android version

Android 16

### Affected phone model

Pixel 8a

### Affected node model

All

### Affected node firmware version

2.6.11

### Steps to reproduce the bug

1. Have a meshtastic device
2. Select one in the Bluetooth menu

During the time it connects, it will not highlight its bullet. This is more problematic when you own multiple devices, as it doesn't show which device it's connecting to until it's got all info - which when a device has hit its maximum NodeDB size of 80 nodes, can be a decent amount of time.

### Actual behavior

Radio button is not filled while loading

### Expected behavior

Radio button should be filled, and ideally also have a way to cancel it like the previous app version did

### Screenshots/Screen recordings

https://github.com/user-attachments/assets/b3a34342-52a7-4a6f-9858-f8aef64382ea

### Relevant log output

```shell
N/A
```

jamesarich, 2025-10-30:
This is resolved in 2.7.5, the status string no longer shows the incorrect device. The device name was removed from the status.

TapGhoul, 2025-11-01:
Forced an update to 2.7.5 real quick - I can confirm the issue is still here.

In the screenshot below, there is nothing showing which device is currently connected, and will not show this info until the device's configuration and node DB has been fully received. This issue has not been resolved.

[image]

TapGhoul, 2025-11-01:
> I see, you want to be able to `abandon` a connection and expect the app (and node) to clean it's state up appropriately when you do.
>
> Looking forward to seeing this implementation.

Or even just display what device it is connecting to, as the app had done before the UI was changed as of 2.7.x.

This used to be standard functionality. Connections can be slow and I don't understand why we can't at least just display what device is currently being connected to.

TapGhoul, 2025-11-01:
Some notes from hacking around today: The connection state is actually correct when it should be, the issue seems to be UI refresh.
When tracing, `MeshService::onConnectionChanged()` and `MeshService::updateServiceStatusNotification()` do get called as soon as the bluetooth connection is established and are aware that the state is now `ConnectionState::CONNECTED`,, which *should* have resulted in the UI bullet being highlighted prior to the connection status being visible.

However, for whatever reason, the UI is then refreshing with the old state, believing it is still disconnected. I'm not fully sure why, I'm not familiar enough with jetpack compose or kotlin as a whole, but it feels like some kind of update throttling or other behavior leaving it showing an incorrect state.

When these change, `ConnectionsScreen` does actually get a call, however, the `connectionState` coming from `connectionsViewModel.connectionState` seems to report a stale value - it says `ConnectionState::DISCONNECTED` when the service should have already broadcasted out that it is in fact `ConnectionState::CONNECTED` - so something is getting lost in handing off to the UI, resulting in an incorrect refresh.

TapGhoul, 2025-12-02:
This has been fixed for the most part in the latest version

closed 2025-12-02 by TapGhoul